puzzlepaint/camera_calibration #4: Having trouble with running calibration script

mostafarohani, 2020-02-05:
Hello, I am able to successfully run the feature detection for 2 cameras I have
`
(lab42) root@gpu2:/nfs1/mostafa/camera_calibration/build# ./applications/camera_calibration/camera_calibration     --pattern_files /nfs1/mostafa/custom_pattern.yaml     --image_directories /nfs1/mostafa/cali$
ration_images/4103636377,/nfs1/mostafa/calibration_images/4103642089 --dataset_output_path /nfs1/mostafa/fea
tures_10px.bin --refinement_window_half_extent 10

time                            file:line     v|
23:27:47.735              loguru.cpp:587   INFO| arguments: ./applications/camera_calibration/camera_calibration --pattern_files /nfs1/mostafa/custom_pattern.yaml --image_directories /nfs1/mostafa/calibratio$
_images/4103636377,/nfs1/mostafa/calibration_images/4103642089 --dataset_output_path /nfs1/mostafa/features_10px.bin --refinement_window_half_extent 10
23:27:47.735              loguru.cpp:590   INFO| Current dir: /nfs1/mostafa/camera_calibration/build
23:27:47.735              loguru.cpp:592   INFO| stderr verbosity: 2
23:27:47.735              loguru.cpp:593   INFO| -----------------------------------
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
23:27:48.539          calibration.cc:1175  INFO| Found 15 images to extract features from.
23:27:48.786          calibration.cc:1210  INFO| /nfs1/mostafa/calibration_images/4103636377/0.png: 52 features
23:27:48.824          calibration.cc:1210  INFO| /nfs1/mostafa/calibration_images/4103636377/1.png: 52 features
23:27:48.868          calibration.cc:1210  INFO| /nfs1/mostafa/calibration_images/4103636377/10.png: 50 features
23:27:48.911          calibration.cc:1210  INFO| /nfs1/mostafa/calibration_images/4103636377/11.png: 52 features
23:27:48.954          calibration.cc:1210  INFO| /nfs1/mostafa/calibration_images/4103636377/12.png: 52 features
23:27:49.002          calibration.cc:1210  INFO| /nfs1/mostafa/calibration_images/4103636377/13.png: 52 features
23:27:49.052          calibration.cc:1210  INFO| /nfs1/mostafa/calibration_images/4103636377/14.png: 52 features
23:27:49.097          calibration.cc:1210  INFO| /nfs1/mostafa/calibration_images/4103636377/2.png: 50 features
23:27:49.138          calibration.cc:1210  INFO| /nfs1/mostafa/calibration_images/4103636377/3.png: 52 features
23:27:49.174 feature_detector_tagged:729   INFO| Timing
------
DetectFeatures()                      10        000000.4291     (00.042912 +- 00.060850)        [00.020548,00.216021]
AprilTag detection                    10        000000.1098     (00.010982 +- 00.001589)        [00.008832,00.013215]
PredictAndDetectFeatures()            10        000000.1045     (00.010452 +- 00.001626)        [00.008745,00.013060]
RefineFeatureDetections()             48        000000.0883     (00.001841 +- 00.000373)        [00.000539,00.002531]

23:27:49.174          calibration.cc:1210  INFO| /nfs1/mostafa/calibration_images/4103636377/4.png: 46 features
23:27:49.218          calibration.cc:1210  INFO| /nfs1/mostafa/calibration_images/4103636377/5.png: 52 features
23:27:49.257          calibration.cc:1210  INFO| /nfs1/mostafa/calibration_images/4103636377/6.png: 21 features
23:27:49.294          calibration.cc:1210  INFO| /nfs1/mostafa/calibration_images/4103636377/7.png: 51 features
23:27:49.340          calibration.cc:1210  INFO| /nfs1/mostafa/calibration_images/4103636377/8.png: 52 features
23:27:49.388          calibration.cc:1210  INFO| /nfs1/mostafa/calibration_images/4103636377/9.png: 50 features
23:27:49.901          calibration.cc:1210  INFO| /nfs1/mostafa/calibration_images/4103642089/0.png: 51 features
23:27:50.199          calibration.cc:1210  INFO| /nfs1/mostafa/calibration_images/4103642089/1.png: 50 features
23:27:50.347          calibration.cc:1210  INFO| /nfs1/mostafa/calibration_images/4103642089/10.png: 52 features
23:27:50.625          calibration.cc:1210  INFO| /nfs1/mostafa/calibration_images/4103642089/11.png: 50 features
23:27:50.833 feature_detector_tagged:729   INFO| Timing
------
DetectFeatures()                      20        000000.6796     (00.033978 +- 00.042980)        [00.020548,00.216021]
AprilTag detection                    20        000000.2188     (00.010939 +- 00.002025)        [00.008742,00.015323]
PredictAndDetectFeatures()            20        000000.2096     (00.010478 +- 00.001681)        [00.008584,00.013676]
RefineFeatureDetections()             94        000000.1740     (00.001852 +- 00.000359)        [00.000539,00.002600]

23:27:50.833          calibration.cc:1210  INFO| /nfs1/mostafa/calibration_images/4103642089/12.png: 52 features
23:27:50.969          calibration.cc:1210  INFO| /nfs1/mostafa/calibration_images/4103642089/13.png: 52 features
23:27:51.172          calibration.cc:1210  INFO| /nfs1/mostafa/calibration_images/4103642089/14.png: 52 features
23:27:51.268          calibration.cc:1210  INFO| /nfs1/mostafa/calibration_images/4103642089/2.png: 43 features
23:27:51.498          calibration.cc:1210  INFO| /nfs1/mostafa/calibration_images/4103642089/3.png: 51 features
23:27:51.780          calibration.cc:1210  INFO| /nfs1/mostafa/calibration_images/4103642089/4.png: 49 features
23:27:51.918          calibration.cc:1210  INFO| /nfs1/mostafa/calibration_images/4103642089/5.png: 49 features
23:27:52.049          calibration.cc:1210  INFO| /nfs1/mostafa/calibration_images/4103642089/6.png: 39 features
23:27:52.207          calibration.cc:1210  INFO| /nfs1/mostafa/calibration_images/4103642089/7.png: 50 features
23:27:52.403          calibration.cc:1210  INFO| /nfs1/mostafa/calibration_images/4103642089/8.png: 51 features
23:27:52.808 feature_detector_tagged:729   INFO| Timing
------
DetectFeatures()                      30        000000.9500     (00.031666 +- 00.034993)        [00.020548,00.216021]
AprilTag detection                    30        000000.3433     (00.011443 +- 00.002303)        [00.008742,00.017321]
PredictAndDetectFeatures()            30        000000.3172     (00.010573 +- 00.001670)        [00.008584,00.013676]
RefineFeatureDetections()            143        000000.2658     (00.001859 +- 00.000379)        [00.000539,00.002633]

23:27:52.808          calibration.cc:1210  INFO| /nfs1/mostafa/calibration_images/4103642089/9.png: 52 features
23:27:52.843              loguru.cpp:467   INFO| atexit
`

however, when i actually try to run the calibration I get this issue,
```
lab42) root@gpu2:/nfs1/mostafa/camera_calibration/build# ./applications/camera_calibration/camera_calibration --dataset_files /nfs1/mostafa/features_10px.bin --output_directory /nfs1/mostafa/result_10px_central_opencv_50 --cell_length_in_pixels 50 --model central_generic --num_pyramid_levels 4
time                            file:line     v|
00:49:50.136              loguru.cpp:587   INFO| arguments: ./applications/camera_calibration/camera_calibration --dataset_files /nfs1/mostafa/features_10px.bin --output_directory /nfs1/mostafa/result_10px_central_opencv_50 --cell_length_in_pixels 50 --model central_generic --num_pyramid_levels 4
00:49:50.136              loguru.cpp:590   INFO| Current dir: /nfs1/mostafa/camera_calibration/build
00:49:50.136              loguru.cpp:592   INFO| stderr verbosity: 2
00:49:50.136              loguru.cpp:593   INFO| -----------------------------------
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
00:49:50.980          calibration.cc:1268  INFO| Dataset 0: /nfs1/mostafa/features_10px.bin
00:49:50.981       calibration_io.cc:244   INFO| Loaded dataset with 1479 feature observations.
00:49:50.981          calibration.cc:937   INFO| Calibrate() starting ...
00:49:50.981          calibration.cc:951   INFO| Estimating the dense initialization for camera 0 ...
00:49:54.875 dense_initialization.cc:1310  INFO| Final init_indices: 12, 7, 1
00:49:54.875 dense_initialization.cc:1311  INFO| best_num_point_triples: 3757
00:49:54.875 dense_initialization.cc:997   INFO| optical_center: -0.0761606   0.166054   -1.52539
00:49:54.875 dense_initialization.cc:998   INFO| init_known_geometry_index: 0
00:49:54.875 dense_initialization.cc:1002  INFO| Determining initial partial calibration from initial images ...
00:49:54.878 dense_initialization.cc:1040  INFO| Extending initial calibration by localizing the remaining images ...
double free or corruption (out)
Aborted
```

Do you have any idea what is going on or what I should do?

puzzlepaint, 2020-02-05:
To pin down the issue, it would be helpful to run the application in `gdb` and/or `valgrind` (while ensuring that it is built in either RelWithDebInfo or Debug mode). For example, run:
```
gdb --ex run --args ./applications/camera_calibration/camera_calibration --dataset_files /nfs1/mostafa/features_10px.bin --output_directory /nfs1/mostafa/result_10px_central_opencv_50 --cell_length_in_pixels 50 --model central_generic --num_pyramid_levels 4
```
Once it crashes, type `bt` and press return to get a backtrace.

painterdrown, 2020-04-03:
Hey, I just got the same "double free or corruption" trouble. And I tried to use gdb and it showed:
[image]
[image]

puzzlepaint, 2020-04-03:
Thanks for the gdb output. I had similar issues to this before when the compile flags between my application and OpenGV differed. I suspect that this might lead to a different memory layout for objects from the Eigen library, which then causes invalid accesses.

Is there any chance that this might be the case in your environment? In case you compiled OpenGV yourself with default settings, it should work, I believe. But in case it has been configured differently, e.g., as part of a system-installed package, maybe it could fail due to such an inconsistency.

The relevant compiler flag was `-march=native`. By default this should be set by both the camera calibration application and in OpenGV. You could try removing this flag for the camera calibration application (in its CMakeLists.txt files, both in the root directory of the sources and in applications/camera_calibration/CMakeLists.txt in all places where it appears) and see whether it helps.

Regarding the negative values in sqrt, I have seen those as well but I don't know what exactly causes them. This indicates some error in the initialization part, however, this should usually not matter, since the initialization is attempted many times with different inputs, and the best result is chosen. So it should not matter if it fails a few times.

painterdrown, 2020-04-04:
Thanks for your advice. I tried remove all `-march=native` flags in both CMakeLists.txt files. But it turned out another error (still related to memory problem, I guess):
[image]
[image]

puzzlepaint, 2020-04-04:
Okay, seems like the flag might not be the problem. But I believe that it is likely to be some inconsistency of that kind. For example, another possibility would be that OpenGV and the camera calibration application could be compiled against different versions of Eigen. But I can only guess the exact reason.

(By the way, it is possible to paste command-line output as text here, preferably delimited by ``` for nice formatting; there is no need to attach it as images.)

painterdrown, 2020-04-04:
Sorry about the images, I have to do so because I am working on a remote server with no display while the application requires GUI. I hate those images too😷.

So next step should I check the exact version of OpenGV and Eigen?

puzzlepaint, 2020-04-04:
That is at least what I would try to check if I got these errors. All used library versions and compiler flags should be identical for compiling OpenGV and for compiling the camera calibration application, otherwise there might be the possibility of incompatible differences.

Another thing to try is to run the OpenGV test that is among the unit tests. If that crashes, it is a pretty clear sign that there is indeed some kind of inconsistency in the compiling that causes the error. If it does not crash, maybe the error is somewhere else. It should be possible to run this test as follows after compiling `camera_calibration_test`:

```
./applications/camera_calibration/camera_calibration_test --gtest_filter=OpenGV.AbsolutePoseSacProblem
```

lingbo-yu, 2020-06-12:
hi, @puzzlepaint :
**I got the same problem like :**

>*** Error in `/home/lingbo/cam-calib-ws/camera_calibration/build/applications/camera_calibration/camera_calibration': double free or corruption (out): 0x00007fffd0001600 ***
> 
> ......
> 
> Thread 4 "camera_calibrat" received signal SIGABRT, Aborted.
> [Switching to Thread 0x7fffe13bd700 (LWP 28954)]
> 0x00007ffff514b428 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
> 54 ../sysdeps/unix/sysv/linux/raise.c: No such file or directory.

**when I run with:**

`sudo gdb --ex run --args ./camera_calibration/build/applications/camera_calibration/camera_calibration --dataset_files ./features_15px.bin --output_directory ./result_15px_central_opencv_50 --cell_length_in_pixels 50 --model central_generic --num_pyramid_levels 4 --show_visualizations`

**I am checking the version of the dependency, or is there a way to solve it ?  thks**

lingbo-yu, 2020-06-12:
I solved this problem. After recompiling with the same version of Eigen, I solved the above problem.