puzzlepaint/camera_calibration #5: Unable to find features

eschneider1992, 2020-02-10:
Hi, I'm excited to try this calibration out but am having problems on the "Calibrating a camera from images in a folder" section of the readme. A CUDA-capable graphics card is not available, so I unchecked the "Live feature detection" button in the capture GUI and got ~150 images of `pattern_resolution_17x24_segments_16_apriltag_0.yaml`. The only notable thing I can think of in the feature-finding command is that `--no_cuda_feature_detection` is turned on:
```
${CALIBRATION_PATH}/build/applications/camera_calibration/camera_calibration \
    --pattern_files ${CALIBRATION_PATH}/applications/camera_calibration/patterns/pattern_resolution_17x24_segments_16_apriltag_0.yaml \
    --image_directories ${DATASET}/images \
    --dataset_output_path ${DATASET}/features_${HALF_WINDOW_SIZE}px.bin \
    --refinement_window_half_extent ${HALF_WINDOW_SIZE} \
    --show_visualizations \
    --no_cuda_feature_detection
```

I tried `HALF_WINDOW_SIZE` values of [5, 10, 15... 35], with the same result. There's a lot of this:
```
14:50:25.615          calibration.cc:1210  INFO| camera_calib_images/images/000084.png: 0 features
14:50:25.903          calibration.cc:1210  INFO| camera_calib_images/images/000085.png: 0 features
14:50:26.151          calibration.cc:1210  INFO| camera_calib_images/images/000086.png: 0 features
14:50:26.400          calibration.cc:1210  INFO| camera_calib_images/images/000087.png: 0 features
14:50:26.642          calibration.cc:1210  INFO| camera_calib_images/images/000088.png: 0 features
14:50:26.875 feature_detector_tagged:729   INFO| Timing
------
DetectFeatures()                  90    000016.8805 (00.187562 +- 00.020580)    [00.125142,00.261750]
PredictAndDetectFeatures()        90    000012.9827 (00.144252 +- 00.018448)    [00.086499,00.189654]
RefineFeatureDetections()         90    000012.9821 (00.144246 +- 00.018448)    [00.086494,00.189649]
AprilTag detection                90    000001.8834 (00.020926 +- 00.002415)    [00.015397,00.026967]

14:50:26.875          calibration.cc:1210  INFO| camera_calib_images/images/000089.png: 0 features
14:50:27.112          calibration.cc:1210  INFO| camera_calib_images/images/000090.png: 0 features
14:50:27.363          calibration.cc:1210  INFO| camera_calib_images/images/000091.png: 0 features
14:50:27.599          calibration.cc:1210  INFO| camera_calib_images/images/000092.png: 0 features
14:50:27.820          calibration.cc:1210  INFO| camera_calib_images/images/000093.png: 0 features
```

The images seem reasonable to me and similar to https://github.com/puzzlepaint/camera_calibration/issues/1. Could blurriness be an issue? I'm holding the calibration board by hand, perhaps there could be a slight jitter there that would have an effect? The pictures look reasonably sharp by eye.

[image]

[image]

puzzlepaint, 2020-02-10:
The CUDA version of the feature detection worked fine on your images, but the CPU version failed. It seems that I forgot to update the CPU version of the feature detection at some point, sorry for that! I changed a single line in the code to match the CUDA version (commit 2587997) which seems to have fixed it. The results of the CPU version on the synthetic unit test now look fine and I also get results for the two images you provided now (with `--refinement_window_half_extent 15`).

eschneider1992, 2020-02-11:
That did it, thanks!

closed 2020-02-11 by eschneider1992