puzzlepaint/camera_calibration #6: Calibration depth to RGB in RealSense 415

swtyree, 2020-02-17:
Hi, are there any special considerations for calibrating the extrinsics between the depth output and RGB camera of a RealSense 415 (for aligning depth to RGB)? Also, is there any reason I shouldn’t expect the code to work for the 415 model? Thanks!

puzzlepaint, 2020-02-17:
Do you plan to calibrate the extrinsics only and use the original depth intrinsics of the camera? Unless you force to use the original intrinsics during the new calibration (which could be attempted with the `--localize_only` parameter), I think that this might potentially be problematic, as it is probably best to use extrinsics and intrinsics that were calibrated together. For example, in a given calibration, a small rotation in the extrinsics may perhaps be compensated by a corresponding change in the intrinsics in the opposite direction. If such variance is possible given the used calibration model, and it causes differences between the original and the new calibration, then mixing the old intrinsics with the new extrinsics might not work as well as expected. I am not sure though whether that would be relevant, just something to keep in mind.

I hope that the code should work for the D415 model, with the main difference to the D435 model being that recording "clean" calibration data for the infrared cameras requires to record still images.

swtyree, 2020-02-18:
Thanks for the information! I'll let you know if I have further questions.

closed 2021-01-26 by puzzlepaint