puzzlepaint/camera_calibration #7: Can computing depth images be done without a GPU?

eschneider1992, 2020-03-06:
Feature detection, for example, can be done without using the GPU using the `--no_cuda_feature_detection` flag. Can something similar be done for depth image computation? I didn't spot any relevant arguments in the help text.

Here's the error that depth computation is getting at the moment:
```
camera_calibration $ ${CALIBRATION_PATH}/build/applications/camera_calibration/camera_calibration     --stereo_depth_estimation     --state_directory ${CALIBRATION_RESULT}     --images ${STEREO_DATASET}/images0/${IMAGE}.png,${STEREO_DATASET}/images1/${IMAGE}.png     --output_directory ${STEREO_DATASET}/stereo_${IMAGE}
time                            file:line     v| 
18:49:59.607              loguru.cpp:587   INFO| arguments: /home/user/camera_calibration/build/applications/camera_calibration/camera_calibration --stereo_depth_estimation --state_directory /home/user/Desktop/camera_calib_images/result_15px_central_generic_20 --images /home/user/Desktop/camera_calib_images/images0/000718.png,/home/user/Desktop/camera_calib_images/images1/000718.png --output_directory /home/user/Desktop/camera_calib_images/stereo_000718
18:49:59.607              loguru.cpp:590   INFO| Current dir: /home/user/camera_calibration
18:49:59.607              loguru.cpp:592   INFO| stderr verbosity: 2
18:49:59.607              loguru.cpp:593   INFO| -----------------------------------
18:49:59.851       calibration_io.cc:696   INFO| Loaded type: CentralGenericModel
18:49:59.936       calibration_io.cc:696   INFO| Loaded type: CentralGenericModel
Stack trace:
7       0x7f679294541d clone + 109
6       0x7f678d0de6ba /lib/x86_64-linux-gnu/libpthread.so.0(+0x76ba) [0x7f678d0de6ba]
5       0x7f67931dfc80 /usr/lib/x86_64-linux-gnu/libstdc++.so.6(+0xb8c80) [0x7f67931dfc80]
4       0x7f67945d5f6e /home/user/camera_calibration/build/liblibvis.so(+0x86f6e) [0x7f67945d5f6e]
3       0x7f6794e08e3d vis::StereoDepthEstimation(std::string const&, std::vector<std::string> const&, std::string const&) + 7309
2       0x7f6792213c4d vis::PatchMatchStereoCUDA::ComputeDepthMap(vis::CUDAUnprojectionLookup2D const&, vis::Image<unsigned char> const&, Sophus::SE3<float, 0> const&, vis::PixelCornerProjector const&, vis::Image<unsigned char> const&, vis::Image<unsigned char> const&, Sophus::SE3<float, 0> const&, vis::Image<float>*, vis::Image<float>*) + 2845
1       0x7f6792227ac9 vis::InitPatchMatchCUDA(vis::StereoParametersSingle const&, int, float, float, float, vis::CUDABuffer_<curandStateXORWOW>*, float, vis::CUDABuffer_<float>*) + 4153
0       0x7f6794331d73 loguru::StreamLogger::~StreamLogger() + 163
18:50:00.328 patch_match_stereo_kern:102   FATL| Cuda Error: invalid device symbol
Aborted (core dumped)
```

puzzlepaint, 2020-03-06:
There is no implementation of depth image computation on the CPU in the camera calibration application. While it should be relatively straightforward to port the CUDA implementation to the CPU, it is pretty unoptimized and thus it is already very slow on GPUs, so it would be expected to be extremely slow on CPUs. So, not sure whether it would be useful at all.

There is an old version of an incomplete CPU-based patch match stereo implementation in `libvis/src/libvis/patch_match_stereo.cc / h`. If you are interested in creating a CPU implementation, it might be easier to start from the current CUDA implementation though.

eschneider1992, 2020-03-06:
That was pretty much what I was guessing, thanks for the answer!

closed 2020-03-06 by eschneider1992