puzzlepaint/camera_calibration #8: Windows build - CUDA issue

yrzhanov, 2020-03-06:
I have NVIDIA card only on Windows 10 machine, so I tried to build the projects on it. The build is mostly successful (with a few tweaks that I am happy to report). All the libraries compile fine, but the application gives linking errors of type:
error LNK2019: unresolved external symbol __cudaRegisterLinkedBinary_69_tmpxft_00007b48_00000000_7_patch_match_stereo_kernel_mutation_cpp
which indicate that all CUDA objects (except for patch_match_stereo_kernel_optimize.cu, which is empty) are not linked. However, libvis_cuda.lib is being explicitly linked. 
I know that the testing has been done on Linux, but maybe someone has any idea what is going wrong in my case?

puzzlepaint, 2020-03-08:
While I don't know for sure what the issue is here, I would suspect that it has something to do with separable linking of CUDA code. Googling something like "visual studio cudaRegisterLinkedBinary link error" should yield a number of possible fixes. Here is one of them:

https://stackoverflow.com/questions/50033435/cmake-cuda-separate-compilation-static-lib-link-error-on-windows-but-not-on-ubun

urotanke95, 2020-07-06:
I'm also trying to build by Windows, because of the similar reason. I'm sure there's a lot of rewriting for headers, etc., it would be great if you could share the whole project which was built successfully.

Andyjk1406, 2021-01-07:
I agree. Straight off the CMake build fails in Windows. It would be really useful to know if anyone got it to build (with or without CUDA).
I'm using vcpkg for most of the dependencies, although opengv was built separately with no issues.