puzzlepaint/camera_calibration #9: Creating undistorted images

FranzEricSchneider, 2020-03-24:
Hi, because depth computation doesn't work without the GPU, and I'd like to try my own stereo algorithm on images, I was wondering if it would be possible to get corrected/undistorted images using this calibration and feed those into a stereo algorithm. Is that something you've tried before? It feels like some of the pieces are available in the "How to use generic camera models in your application" section of the README, although it's not a 1:1 fit.

https://github.com/puzzlepaint/camera_calibration/blob/master/applications/camera_calibration/generic_models/src/central_generic.h
```
  /// Un-projects the given image pixel coordinate to a 3D direction in the
  /// local coordinate frame of the camera.
```

This would be something new for me, and I imagine I could dig up some online resources on it, but first I wanted to check if it was something your team had tried before. Thanks!

puzzlepaint, 2020-03-25:
I haven't implemented this. However, as long as a central model is used, it should not be difficult to undistort the images with the model in order to use them with other software that expects pinhole images, or to perform stereo rectification for a pair of images if that is desired. I think this would be a nice addition to the camera class linked above.

puzzlepaint, 2020-03-25:
For an existing implementation of undistortion and stereo rectification, see for example:
https://github.com/colmap/colmap/blob/dev/src/base/undistortion.cc