puzzlepaint/camera_calibration #10: What is the physical meaning of unit of translation parameter?

ru040tp6, 2020-04-21:
Hi, 
What is the physical meaning of unit of translation parameter?
(tx,ty,tz in the camera_tr_rig.yaml)

Does it measure by the calibration pattern size? (eg: 13mm)
Could you help me answer this question?
Thanks

puzzlepaint, 2020-04-21:
Yes, the scale of the translation is derived from the given calibration pattern size. However, unless you manufacture the pattern very precisely, I would suggest to treat that as an approximate value.

ru040tp6, 2020-04-22:
Hi Thomas,

I have used *pattern_resolution_17x24_segments_16_apriltag_0.pdf* to
calibrate my stereo camera
I want to check the extrinsic parameter whether is reasonable.

I projected the pattern on the screen size, and the actual size is noted on
attach image.
The stereo camera baseline is 65mm.
The output *camera_tr_rig.yaml * are listed as following.

How to check the extrinsic parameter approximately?

Thanks for your help.

pose_count: 2
poses:
  - index: 0
    tx: -0.001510846769417
    ty: -0.00024836950728927
    tz: -0.0010183190599437
    qx: -0.00070368217318108
    qy: 0.0024848476276176
    qz: -0.0031055784802162
    qw: 0.99999184283972
  - index: 1
    tx: -0.03022234772154
    ty: -0.00025584277402164
    tz: 0.00034163303126077
    qx: -0.01043780363897
    qy: -0.0088663911147729
    qz: -0.0011882905212975
    qw: 0.99990550920046

 example.png
<https://drive.google.com/file/d/1Dae91etXE9K_SIlsstCaiqYqJMUGxKJc/view?usp=drive_web>

puzzlepaint, 2020-04-22:
How to check this:

* The poses in camera_tr_rig.yaml are the "image_tr_pattern" poses. Invert them to get the "pattern_tr_image" poses. The resulting translation components represent the locations of the projection centers relative to some arbitrary pattern coordinate system. Subtract them from each other and take the norm of that difference vector to obtain the computed baseline.
* Measure the actual distance between two adjacent features on the pattern and compare this to the square_length_in_meters setting in the .yaml file corresponding to the pattern you used. Scale the "ground truth" baseline of 65mm accordingly and compare the scaled baseline to the baseline computed above.

(The example.png on Google Drive is not accessible, it says that permission is required.)

ru040tp6, 2020-04-23:
Hi Thomas,

Many thanks for your detailed description.
I checked the results, and it seems ok.
I also summarized the procedure as attached image.

Note: the actual baseline is 32 mm, not 65 mm as previously discussed.

 calibration_check.png
<https://drive.google.com/file/d/1JoJkKnD17pdFuSoNqE5PSX9i-lVTHe3Q/view?usp=drive_web>

closed 2020-04-23 by ru040tp6