puzzlepaint/camera_calibration #11: How can I capture the images that the tool can extract features successfully?

ru040tp6, 2020-04-27:
Hi @puzzlepaint,
This calibration tool is well to use, and thanks for your sharing.
However, I found the feature extraction of many images I captured failed, and the message showed 
"0 features".
I have follow the instructions, using global shutter sensors and capturing on the tripod.
Is there any other tips for capturing?
Or what situation the tool would rule out the images for feature extraction?

puzzlepaint, 2020-04-27:
The program will only detect a pattern if its AprilTag is fully within the image. In addition, the pattern should not be too small or too large in the image, and the lighting conditions should be uniform.

These are some general rules to follow. In general, the program is rather careful with extracting features and rather discards a feature than possibly includes an outlier. It is best suited for global-shutter cameras with which videos can be recorded, since then discarding some images does not matter much.

ru040tp6, 2020-04-28:
Thanks for your reply.
I will capture more images to get more features.

closed 2020-04-28 by ru040tp6