puzzlepaint/camera_calibration #12: Red cell on the report_cameraX_error_magnitudes.png

ru040tp6, 2020-04-28:
Hi @puzzlepaint ,

I try to calibrate a ultra-wide lens stereo camera.
The results of [report_camera0_error_magnitudes.png](https://drive.google.com/open?id=1kl8jTOWomn7KgRgo_g_QYeRecNlAT0kr) and [report_camera1_error_magnitudes.png](https://drive.google.com/open?id=1YxgYPzYgrPyfEEj1Ad8Fi64NjLyy0Hp6) have some red cells.
How large is the max error magnitude of these red cells?
And you have mention that "If there are red points forming some systematic pattern, something is probably wrong". 
What is the probable reasons?
How to get more robust or more convincing calibration result?

I provide my [calibration results](https://drive.google.com/open?id=1NAw88MIX7-R1uuw8aUCC4xAn7L0anab3), and maybe you can help me check what is going wrong.

puzzlepaint, 2020-04-28:
Pure red in this visualization is used for reprojection errors >= 0.5 pixels ([see here](https://github.com/puzzlepaint/camera_calibration/blob/2587997018fd08f0325caf1599bee1778c3aeda4/applications/camera_calibration/src/camera_calibration/calibration_report.cc#L772)). There is no maximum for display in this visualization; this value is the minimum.

The maximum reprojection error (*after removing outliers*) is given in the `_info.txt` files. In this case, this is actually very close to 0.5 pixels. The median and average error look good though.

The pattern on the left looks like it might correspond to one or two calibration images only, since it seems to be caused by one or two projections of the pattern only. So, something seems to be more inconsistent for these images than for the others. Some ideas for possible reasons:

* Use of the central camera model with these images being in a pose that strongly exhibits the non-central effect (so it might be worth trying to repeat the calibration with the non-central model to see whether the effect goes away)
* The optimization being unable to optimize the poses of these images for some reason (e.g., due to a bug in the calibration program, or due to not converging within the iteration limit)
* Motion blur in these images making the detection hard
* Rolling shutter images of a moving pattern introducing distortions
* The pattern not being static, but being slightly bent in these images
* The camera rig not being static, but being slightly bent when taking these images
* Discontinuities in the lens that cannot be modeled, or cannot be modeled sufficiently well, since the camera model assumes some smoothness (could possibly be addressed by using a higher grid resolution to some extent, but in case the problem is actually something else, this might simply hide the problem without fixing it unless more calibration data is recorded to match the new resolution)

There are surely more possible reasons that I don't think of right now.

ru040tp6, 2020-04-29:
Thanks for your detailed explanation.
I use the non-central model, and the error seems better.
([report_camera0_error_magnitudes.png,](https://drive.google.com/open?id=163uvDxEQ1nd2juBrostwOte6EsKEyIwB) [report_camera1_error_magnitudes.png](https://drive.google.com/open?id=1hg2qxNUZ8mkKJfRfe_FPftMWfxm6Ole3))

closed 2020-04-29 by ru040tp6