puzzlepaint/camera_calibration #16: Is it possible to provide some open datasets for test？

lingbo-yu, 2020-06-04:


puzzlepaint, 2020-06-11:
I uploaded some sample images here (roughly 1 GB in size):
https://drive.google.com/file/d/1F4bemXktcBqAH0stcCPJkhfqIZrQ4b6u/view?usp=sharing

They were recorded with an infrared camera of an Intel D435 depth camera. The corresponding pattern description file should be `pattern_resolution_17x24_segments_16_apriltag_0.yaml`.

lingbo-yu, 2020-06-12:
@puzzlepaint:
This is really the best news of the day! thank you very much!

lingbo-yu, 2020-06-12:
hi, @puzzlepaint : 
 **I got the same problem like :** 

> *** Error in `/home/lingbo/cam-calib-ws/camera_calibration/build/applications/camera_calibration/camera_calibration': double free or corruption (out): 0x00007fffd0001600 ***
......
Thread 4 "camera_calibrat" received signal SIGABRT, Aborted.
[Switching to Thread 0x7fffe13bd700 (LWP 28954)]
0x00007ffff514b428 in __GI_raise (sig=sig@entry=6) at ../sysdeps/unix/sysv/linux/raise.c:54
54	../sysdeps/unix/sysv/linux/raise.c: No such file or directory.

**when I run with:**

`sudo gdb --ex run --args ./camera_calibration/build/applications/camera_calibration/camera_calibration --dataset_files ./features_15px.bin --output_directory ./result_15px_central_opencv_50 --cell_length_in_pixels 50 --model central_generic --num_pyramid_levels 4 --show_visualizations`

closed 2021-01-26 by puzzlepaint

Dorothy-2016, 2021-07-27:
@puzzlepaint  Hi, I test the data which you uploaded before with central model and noncentral model, then test the params using generic_models example, but it always output "Unprojection failed! Pixel: (0.5 0.5)", I don't find which part that I did wrong, do you have  any ideas?

Dorothy-2016, 2021-07-27:
> @puzzlepaint Hi, I test the data which you uploaded before with central model and noncentral model, then test the params using generic_models example, but it always output "Unprojection failed! Pixel: (0.5 0.5)", I don't find which part that I did wrong, do you have any ideas?
> 
> here is central params:
> type : CentralGenericModel
> width : 1280
> height : 800
> calibration_min_x : 15
> calibration_min_y : 15
> calibration_max_x : 1264
> calibration_max_y : 784
> grid_width : 27
> grid_height : 17
> 
> and here is noncentral params:
> type : NoncentralGenericModel
> width : 1280
> height : 800
> calibration_min_x : 15
> calibration_min_y : 15
> calibration_max_x : 1264
> calibration_max_y : 784
> grid_width : 27
> grid_height : 17
> 
> from the calib result, they are the same, but grid is different which is so long that I haven't copied here yet.

Maybe I have found the problem and can solve it. thanks!

spdfghi, 2024-10-19:
> I uploaded some sample images here (roughly 1 GB in size): https://drive.google.com/file/d/1F4bemXktcBqAH0stcCPJkhfqIZrQ4b6u/view?usp=sharing
>
> They were recorded with an infrared camera of an Intel [...]

@puzzlepaint Would you mind to provide Intel D435 RGB too? We have some problem with RGB camera.