puzzlepaint/camera_calibration #17: Difficult to extract featrues on L515 infrared camera, need help.

cdb0y511, 2020-07-09:
Hi,@puzzlepaint.  Your great work works fine with D435i both infrared and RGB camera on multiple resolutions.
However, it seems very difficult to calibrate the L515 infrared camera cause it has very low resolution and low image quality.
It detects very few of features.  It can see the april tag ,but can not detect the cross line features. Maybe it needs another kind of feature specified designed for infrared images.
Any advice will be helpful. Thanks  
[image]
[image]
[image]

puzzlepaint, 2020-07-11:
Hi, I don't know whether that would work, but this is what I would try:

* Try to use a large window for feature refinement (`--refinement_window_half_extent`) since the blur washes out the features, so a large detection region seems to be required.
* Try generating custom calibration targets with `scripts/create_calibration_pattern.py`.
  * Potentially use less star segments (`--num_star_segments`) such that the feature centers potentially appear clearer. Maybe even a standard checkerboard might perform better than the star pattern in this case.
  * Potentially make the features larger (`--approx_square_length_in_cm`) to improve their visibility.

If none of that helps, one might have to dig into the feature detection code to see whether there is something else that prevents the features from being found correctly, or whether they are discarded for not being accurate enough (or use a different feature detector).

cdb0y511, 2020-07-12:
Thanks. if I have any progress, I will let you know.

cdb0y511, 2020-07-14:
Hi,@puzzlepaint
1. I set half window as 20, and cell as 20px too.
2. --num_star_segments  is still 16, I think star feature is more robust than checkerboard in any circumstance.
3. --approx_square_length_in_cm 1.6. --apriltag_length_in_squares 2
 Detection finally works .
However, the next problem  is: bundle adjustment is not converged.
I wonder if it s the feature detection goes wrong cause the dateset you given converge very fast.
https://drive.google.com/file/d/1F4bemXktcBqAH0stcCPJkhfqIZrQ4b6u/view?usp=sharing
Any advice?
[image]
[image]
resolution : 1024 x 768
horizontal_fov : 71.054815241238
vertical_fov : 56.267277300423

num_localized_imagesets : 2374
num_total_imagesets : 2374

reprojection_error_count : 382866
reprojection_error_median : 0.32672810142639
reprojection_error_average : 0.41641372517283
reprojection_error_maximum : 3.9341197601522
median_kl_divergence : 0.18301077143518

reprojection_error_histogram_visualization_half_extent_in_pixels : 0.20000000298023
maximum_error_visualization_maximum_error_in_pixels : 0.5

Thank you.

puzzlepaint, 2020-07-14:
Regarding 2.: Maybe, I don't know. The checkerboard features would be expected to reduce the blurriness at the feature centers though.

Judging from the L515 images and the residual histogram, I would not be sure about whether it fails to converge. The residuals appear to be centered, and given that the input images are rather blurry, maybe it is to be expected that the residuals remain rather large in this case. If so, that result might be fine.

Running for 7 hours on a high-end PC sounds like it should not be necessary. If the optimization works, then I think it should usually perform some large cost reductions in the first few optimization iterations, and then proceed to only make rather insignificant tiny steps. The convergence criterion is however set such that it won't stop even if it only makes tiny steps. So, possibly it makes sense to limit the iteration count (if it behaves as described) or modify the convergence criterion to reduce the runtime. Perhaps also another choice of the `--schur_mode` parameter might speed up the iterations (especially `dense_cuda`). (These observations are mainly from using the central-generic model; I always applied noncentral-generic as a refinement on top of the central-generic result. It might perhaps converge worse when used directly from the start.)

So, I think that sometimes the runtime (with the default convergence criterion) differs by a lot based on whether the optimization continues to do tiny steps for a long time or not. These steps seem often insignificant and could probably be omitted in these cases.

Apart from that, there is (unfortunately) always the possibility of a bug in the optimization, or of a case where the convergence is bad. Regarding the latter: The optimizer used by the camera calibration program applies the `lambda` parameter to the Hessian approximation matrix H in this way in the update equation of the optimization: `H_Modified = H + lambda * Identity_Matrix`. Recently I tried another variant, which is to apply it as follows: `H_Modified = H + lambda * diag(H)`. This seemed to improve the convergence of the optimization in some cases. However, for the cases I tested so far, it did not seem to cause any significant difference in the final result of the optimization. If you still would like to try that (with dense Schur mode), you could modify the corresponding code in lm_optimizer.h to something like the following, set `m_apply_lambda_on_diag` to true, and choose a fixed starting value for lambda (for example, 0.00005):
```c++
          // Add to the diagonal of H according to the Levenberg-Marquardt method.
          if (!m_on_the_fly_block_processing) {
            int diagonal_index = 0;
            for (usize i = 0; i < m_block_diag_H.size(); ++ i) {
              for (int k = 0; k < m_block_diag_H[i].rows(); ++ k) {
                m_block_diag_H[i](k, k) = m_apply_lambda_on_diag ? ((1 + m_lambda) * m_original_diagonal[diagonal_index]) : (m_original_diagonal[diagonal_index] + m_lambda);
                ++ diagonal_index;
              }
            }
            for (int i = 0; i < dense_degrees_of_freedom; ++ i) {
              m_dense_H(i, i) = m_apply_lambda_on_diag ? ((1 + m_lambda) * m_original_diagonal[diagonal_index]) : (m_original_diagonal[diagonal_index] + m_lambda);
              ++ diagonal_index;
            }
            CHECK_EQ(diagonal_index, degrees_of_freedom);
          }
```

cdb0y511, 2020-07-15:
Hi,@puzzlepaint.Very appreciate your answer,
 I am working on improving the L515's intrinsic. I totally agree with your point in your  paper. I think the accuracy of  intrinsic is very important to pose estimation and reconstruction. It seems difficult to eliminate the intrinsic error in the later step of pipeline. Generic  model is much better than simple pinhole, even for a near pinhole camera. So I wish to use Generic  model as default camera model.
I have modified your code as you told. It seems to avoid tiny step iteration successfully.  It may need further investigation though.

There are still some question about Generic  model on L515:

1. I made a bigger target (A3 size ,checkerboard style). 
[image]

And I set outlier_threshold to 0.8(below 1 pixel) cause I think the detection may be few pixel off on some point. It will remove 1/3 to 1/4 point at end of  Removing outlier with reprojection process. I think it will work after several  trials. I expect generic  model will outperform  default  intrinsic in reconstruction.
If you have any advice about calibrate infrared camera of L515, please let me know.

2. The  grid_width and grid_height can be much different with  same optimize parameters but different feature.bin.
like, your d435i data, with 10 half window and 10 cell. It gets 128x80 grids in the end.
However, my data same d435i infrared about 1.23k images, with 10 half window and 10 cell with same target pattern. It gets 72x45 grids in the end. The l515 infrared is about 51x38. 
I think the denser grid in the intrinsic  means better calibration besides the median error.
It seems to decide by:
  *resolution_x = calibration_area_width / approx_pixels_per_cell + 0.5f + 2 * exterior_cells_per_side;
  *resolution_y = calibration_area_height / approx_pixels_per_cell + 0.5f + 2 * exterior_cells_per_side;
I think it is related with cell parameter, but why does the D435i get different  grid size?

Thank you.

puzzlepaint, 2020-07-15:
1. I don't think I can give additional generic advice regarding the L515 at the moment since I haven't used a L515 device myself yet.

2. The chosen grid size depends on the size of the image area that gets calibrated. The latter is chosen as an axis-aligned bounding box around all of the feature detections and should ideally only be a little smaller than the image. The D435 images are 1280 x 800 pixels. If features are detected very close to all image borders, then it can be neglected that the calibrated image area will be a little smaller. It will approximately be counteracted by the addition of `2 * exterior_cells_per_side` which adds one cell on each side beyond the calibrated image area. Then, the expected grid resolution with `--cell_length_in_pixels` == `approx_pixels_per_cell` == 10 would be (1280 / 10) x (800 / 10) for the D435 images, which is exactly the 128 x 80 that you observed.
    
    Consequently, the maximum grid size for the L515 images should be approximately 102 x 77 with the same settings. Smaller grid sizes would theoretically be used if there are no feature detections close to the image boundaries, such that the size of the calibrated image area is chosen significantly smaller than the image size.
    
    A higher grid resolution for the same area might not always result in a more accurate calibration. It could also mean that the calibration will locally adapt to some inaccurate feature detections.

closed 2021-01-26 by puzzlepaint