puzzlepaint/camera_calibration #18: Pixel corresponding to the (0.0, 0.0, 1.0) vector

mishakin, 2020-07-28:
Hi, @puzzlepaint 
Thanks for sharing the code, it is of great help. I have just started with using it for my needs, sorry for a potentially stupid question.

**Question:** I've calibrated my large-FOV fisheye camera (it is a mono-camera) using central_generic and got the intrinsics0.yaml and modified the code from https://github.com/puzzlepaint/camera_calibration/tree/master/applications/camera_calibration/generic_models to work with my intinrinsics. As a first test, I projected the vector (0.0, 0.0, 1.0) corresponding to the optical axis of the camera to the image, and it turned out to be projected **exactly** at the image center. However I know for sure, the optical axis ray of this camera doesn't correspond to the center of the image. So my question: did I do some thing wrong and may be got incorrect intrinsics, or it is a feature of the non-parameteric model and I have to determine the optical axis from the intrinsics some other way? Basically, what I got is it a bug or a feature?  

Regards,
Sergey.

puzzlepaint, 2020-07-28:
This is a feature. The central-generic model has an ambiguity where rotating all pixels' observation directions in one direction, and rotating all estimated camera poses by the same amount in the opposite direction leaves the calibration constant. So, the orientation of the directions can be chosen freely. The code does this in `CentralGenericModel::ChooseNiceCameraOrientation()`. One of the things this function does is to make the camera's local direction vector `(0, 0, 1)^T` point forward, defined as the direction at the center of the image. This is where this result comes from.

This (somewhat arbitrary) camera orientation choice is merely used in order to:

* Have an orientation that is similar to what would be expected with a parametric model (potentially making it easier to match different calibrations).
* Make the observation-direction-visualization look nicer, since it takes the axis alignment of the directions into account.

The generic model does not attempt to estimate physical properties of the camera such as the optical axis (except for the optical center that is estimated by the central model). If you need to estimate the optical axis for some reason, it could for example be done on top of the generic calibration.

mishakin, 2020-07-28:
Thomas, thanks a lot for explanation.
I'm closing the issue.

closed 2020-07-28 by mishakin