puzzlepaint/camera_calibration #19: How to use generic model in Colmap?

ghetheaber, 2020-07-29:
Hi, @puzzlepaint 
In the paper Colmap reconstruction is used as an evaluation. 
Do you undistort the images first, then give the undistorted images to Colmap?
Or you integrate the generic-model undistortion and distortion into Colmap? If so, what's the Jacobian? 
Thanks in advance!

puzzlepaint, 2020-07-29:
The reconstructions were first created in Colmap with a camera model that is supported by it. Then, additional bundle adjustment was applied to them by loading them in the camera calibration program. See [tools/bundle_adjustment.cc](https://github.com/puzzlepaint/camera_calibration/blob/master/applications/camera_calibration/src/camera_calibration/tools/bundle_adjustment.cc). This may be called from [here](https://github.com/puzzlepaint/camera_calibration/blob/2682d587e1272b8e2ab845074bdf050cf9248ef0/applications/camera_calibration/src/camera_calibration/main.cc#L424) via the camera calibration executable (undocumented).

ghetheaber, 2020-07-30:
Very creative! Let me have a try.
Thanks a lot.