puzzlepaint/camera_calibration #23: Explain camera parameters

DevAhmedYM2018, 2020-09-17:
Hi,
Can you explain which parameters are (distortion parameters) and which are (intrinsic parameters), such that in Pinhole parameters, there are up to 8 distortion parameters, and 2 focal lengths, and 2 optical centers parameters.?

And then what is the equations needed to un-distort the images, and how to project and un-project using un-distorted images (without distortion parameters)?

puzzlepaint, 2020-09-17:
Which camera model does this question refer to?

DevAhmedYM2018, 2020-09-21:
both central generic model and noncentral generic model

puzzlepaint, 2020-09-22:
I think that the parameters in these models cannot be classified as distortion or intrinsic parameters as in traditional camera models. The parameters in the generic models simply specify the observation directions or lines corresponding to some pixels of the camera.

I wrote down the un-distortion algorithm in this issue (only applicable to the central generic model): #22
After un-distortion, the projection and un-projection is done with the camera model that was chosen as un-distortion target (typically the pinhole camera model without distortion), which is independent from the original camera model. So, if the pinhole model is used, then projection and un-projection on the un-distorted image would directly use the traditional pinhole model's equations for that.