puzzlepaint/camera_calibration #24: Signal: SIGABRT

xunkaixin, 2020-11-20:
hi，
  thank you for your code
  The program crashes during calibration

  ./camera_calibration --dataset_files ./output/features_px.bin --output_directory ./pattern_resolution_17x24/output/noncentral_generic --cell_length_in_pixels 20  --model noncentral_generic  --num_pyramid_levels 4 

[image]

 Can you give me some advice?

puzzlepaint, 2020-11-23:
Looks like it crashes in opengv code. There may be issues sometimes caused by different compiler flags or versions of the Eigen library being used to compile opengv and the final application. You can check whether there is a generic issue of this type by running the `OpenGV.AbsolutePoseSacProblem` unit test (compile `camera_calibration_test` and run `./camera_calibration_test --gtest_filter=OpenGV.AbsolutePoseSacProblem`).

If this test passes, then there is likely wrong input provided by the camera_calibration application to opengv. Did you try enabling the visualizations to potentially get more information?

xunkaixin, 2020-11-24:
thank you,  the problem is the version of the eigen,  I've solved it

closed 2020-11-24 by xunkaixin