puzzlepaint/camera_calibration #25: How to obtain camera internal parameters from calibration results

xunkaixin, 2020-11-25:
hi，
      i have get the calibration results,  How to obtain camera internal parameters from calibration results?  i want to correct the fisheye from calibration results

report_camera0_info.txt:

resolution : 1280 x 720
horizontal_fov : 212.87422947398
vertical_fov : 118.26021874509

num_localized_imagesets : 385
num_total_imagesets : 390

reprojection_error_count : 10743
reprojection_error_median : 0.141347150886
reprojection_error_average : 0.17611839469023
reprojection_error_maximum : 1.1953747539699
median_kl_divergence : 0.46867320578886

reprojection_error_histogram_visualization_half_extent_in_pixels : 0.20000000298023
maximum_error_visualization_maximum_error_in_pixels : 0.5

puzzlepaint, 2020-11-29:
After successful calibration, the calibrated intrinsic camera parameters are stored in the files `intrinsicsX.yaml` in the output folder.

It seems that you are interested in undistorting the images to a pinhole camera model using the calibrated intrinsics. I wrote down an algorithm to achieve this here: https://github.com/puzzlepaint/camera_calibration/issues/22

xunkaixin, 2020-11-30:
algorithm： “Project this direction d to a pixel coordinate p2 using the calibrated camera model”

i have get d， but i dont‘t know how to  projet d to p2 by intrinsicsX.yaml

puzzlepaint, 2020-12-01:
You could for example use the Project() function of the camera model implementation in https://github.com/puzzlepaint/camera_calibration/tree/master/applications/camera_calibration/generic_models . These implementations support loading the camera model from yaml files, and they only depend on the Eigen library. See the section "How to use generic camera models in your application" in the top-level Readme.

closed 2020-12-02 by xunkaixin