puzzlepaint/camera_calibration #26: Windows build errors

orgicus, 2020-12-03:
Hi,

Thank you for sharing the source code publicly. 
It saves a lot of implementation time.

Unfortunately, I'm running into compile errors on Windows (after many issues setting up CMake):
```batch
\camera_calibration\applications\camera_calibrat
ion\third_party\apriltag\common\time_util.h(32,10): fatal error C1083: Cannot open include file: 'sys
/time.h': No such file or directory
camera_calibration\applications\camera_calibrat
ion\third_party\apriltag\common/timeprofile.h(35,10): fatal error C1083: Cannot open include file: 's
ys/time.h': No such file or directory
camera_calibration\applications\camera_calibrat
ion\third_party\apriltag\common\workerpool.c(29,10): fatal error C1083: Cannot open include file: 'pt
hread.h': No such file or directory
...
camera_calibration\applications\camera_calibrat
ion\third_party\apriltag\common\pjpeg.c(452,36): error C2466: cannot allocate an array of constant si
ze 0
\camera_calibration\applications\camera_calibrat
ion\third_party\apriltag\common\pjpeg.c(521,34): error C2057: expected constant expression
camera_calibration\applications\camera_calibrat
ion\third_party\apriltag\common\string_util.h(59,43): error C2061: syntax error: identifier '__attrib
ute__'
...
camera_calibration\applications\camera_calibrat
ion\third_party\apriltag\common\pjpeg.c(521,34): error C2466: cannot allocate an array of constant si
ze 0
```
At first glance they seem to be third party library errors related to unix/windows differences.

Any hints on how to correctly configure and build the project would be greatly appreciated.

Additionally I did manage to build on Windows via WSL2 + CUDA + VcXsrv, however WSL2 doesn't support passing over USB devices from windows. I am hoping I can have an executable on windows to stream as an IP camera on localhost, but as far as I can tell camera_calibration populates the cameras list from USB cameras only ? 

[image]

What else could I try to run camera_calibration ?

remiyoudu, 2020-12-04:
Becuase unix only？

puzzlepaint, 2020-12-04:
Since I never tried to run this program on Windows, it is expected to cause some trouble.

For live input, the program tries to use (up to) three different libraries: librealsense2 (for Intel RealSense cameras), the Structure SDK (for Occipital Structure Core cameras), and video4linux2. Unfortunately, I have no idea how feasible it is to make those work in WSL2. Maybe using pre-recorded images instead of live input could be an alternative to get it running at all and see if it works for you? Pre-recorded images can be used via the command line as described in the Readme.

Regarding the compile errors when trying to compile for Windows directly, indeed it seems that they occur in the third-party dependency apriltag. The program uses a copy of the apriltag code that is in `/applications/camera_calibration/third_party/apriltag/`, however I don't remember making any changes to it, so it might be worth a try to update to a more current version from https://github.com/AprilRobotics/apriltag . Still, the video4linux2 input will probably need to be replaced with something else for Windows.

orgicus, 2020-12-04:
@puzzlepaint Thank you so much for the detailed answer.
I won't pursue building camera_calibration on Windows natively.

I missed the pre-recorded images option: looks like a clear option forward!
I'll give that a go via WSL2.

closed 2020-12-04 by orgicus