puzzlepaint/camera_calibration #27: How to obtain the centre of Distortion?

WJinhai, 2020-12-09:
I read a few papers and said that the distortion center will affect the radial distortion, so I want to ask, is the distortion center calculated separately in the code?

puzzlepaint, 2020-12-09:
Assuming this concerns the generic camera models: The distortion center is not estimated separately for those, instead it (and any other parameters that parametric models handle explicitly) is handled implicitly by how the observation lines/directions are arranged. This allows for much more generic distortion handling.

WJinhai, 2020-12-10:
Parameter-free Radial Distortion Correction with Centre of Distortion Estimation， In this paper ,  The author calculated the center of the distortion  separately

lingbo-yu, 2020-12-10:
hi@puzzlepaint， I also have a question for a long time
Do we really need to firstly undistort the pixel which we want to unproject to a direction? and will it make the unprojection more accurate?

puzzlepaint, 2020-12-11:
@WJinhai This is because they model radial distortion explicitly. Therefore, they need to know this center explicitly to be able to model it accurately. However, a generic model can represent any types of distortion without this explicit modeling (which is the whole point of generic models), so it is not necessary to estimate the distortion center separately for generic models. Of course, it is still possible to estimate a center of distortion based on the result of the generic calibration, if desired, but it will not help with the calibration.

@lingbo-yu Which undistortion step do you refer to? In the generic models that we used, the unprojection process is implemented by interpolating in a B-Spline surface and normalizing the direction, and I am not sure which part of that you consider to be the undistortion?