puzzlepaint/camera_calibration #40: Issue calibrating DSLR images.

anushl9o5, 2021-05-04:
Hey, 
I am using an 18-55 lens, the pictures are taken when the lens is fully zoomed out.
The results after calibration seem weird to me, they are all constrained to one side.

### Dataset
[image]
[image]
[image]
[image]

### Configuration
Half window size - 10 (I couldn't get features on all images for 15)
Cell size - 20 (1620x1080)
Camera Model - Central_generic

### Results
[image]
[image]
[image]
[image]

````
What I observed is, for some reason the calibration script only takes 5 images from the entire dataset. Maybe this is why the results are only on one side?
````

Any ideas and help would be much appreciated!

puzzlepaint, 2021-05-04:
Something clearly went wrong here. It looks like the calibration was done based on very few images only, where the pattern is always on the left side. This would cause the bounding box of all feature detections to be only in this region, which would explain why in the second last visualization in your post, all other areas are black.

Are you sure that a half window size setting of 10 gave feature detections in many images? If yes, then another possible issue is that there might be little pixel overlap of the pattern area among calibration images in which the pattern is on the left side, and images in which the pattern is elsewhere in the image. This could cause the program to fail to localize the other images, such that the calibration is done with the images having the pattern on the left only, even if features were also detected in the other images. This could probably be seen from the log output, which should say "Imagesets used for calibration: X / Y" somewhere. In this case, the coverage of the image area by the pattern would need to be improved, for example, by taking some additional calibration images in which the pattern covers all or most of the image, which will then "connect" the existing images that have the pattern in different image areas. The live feature detection is also very helpful for verifying a good feature coverage.

anushl9o5, 2021-05-05:
Yes the features were getting detected for half window size of 10, 15 was not giving any features at all. I collected some more data and this time it worked much better. Not really sure what caused the issue. But I think you're right about there being no overlap between the views. The newly collected data had more images with better coverage.
Unfortunately I can't use the live feature detection, since I am working remotely. 

Thanks for your inputs!

closed 2021-05-05 by anushl9o5