puzzlepaint/camera_calibration #43: Feature Request: Export Target Deformation Info

abrock, 2021-05-13:
I'm interested in measuring the deformation of a printed target.

Could you export that in a file in the report directory? For example for every point of the target one line
x1 y1 z1 x2 y2 z2
where (x1,y1,z1) are the initial coordinates assuming a perfectly planar target with no deformations and (x2,y2,z2) the estimated positions.

puzzlepaint, 2021-05-16:
I think that there is already a file `points.yaml` among the state files that contains the optimized (deformed) pattern point positions, and it should be simple to compute the original point positions from the pattern definition YAML file. See `FeatureDetectorTaggedPattern::GetCorners()` and `PatternData::IsValidFeatureCoord()` in `/applications/camera_calibration/src/camera_calibration/feature_detection/feature_detector_tagged_pattern.cc/h` for how to obtain the feature positions as integers (without scaling). So I think that the required information is already present and relatively easy to access.

abrock, 2021-05-19:
Thank you for the information. I tried to add the functionality I'd like to the `SavePointsAndIndexMapping` function but FeatureDetectorTaggedPattern::GetCorners is not static and I couldn't figure out where to get an instance from. I ran a debugger with a breakpoint in the contructor of `FeatureDetectorTaggedPattern` and it didn't trigger.
Could you give me a rough idea where to start adding the functionality?
I'd also like to get 3D feature points for every image used.

puzzlepaint, 2021-05-20:
This is where the feature detector is constructed (it is only allocated if needed):
https://github.com/puzzlepaint/camera_calibration/blob/846bb5282a23b68e9fd7f3d240f7c979a016dea6/applications/camera_calibration/src/camera_calibration/main.cc#L462

This then builds the map from feature IDs to positions using `GetCorners()` (if image directories are given):
https://github.com/puzzlepaint/camera_calibration/blob/846bb5282a23b68e9fd7f3d240f7c979a016dea6/applications/camera_calibration/src/camera_calibration/calibration.cc#L1266

It should be possible to use this mapping after this has been called. This map is also saved in the dataset files as the "known geometries":
https://github.com/puzzlepaint/camera_calibration/blob/846bb5282a23b68e9fd7f3d240f7c979a016dea6/applications/camera_calibration/src/camera_calibration/io/calibration_io.cc#L113

It could be an option to load them from these files and then associate them with the optimized point positions based on the feature ID. This should not require modifying the camera calibration code.

The features for each image are also saved in the dataset files:
https://github.com/puzzlepaint/camera_calibration/blob/846bb5282a23b68e9fd7f3d240f7c979a016dea6/applications/camera_calibration/src/camera_calibration/io/calibration_io.cc#L104