puzzlepaint/camera_calibration #45: Docker Support

kudo1026, 2021-06-22:
Hi,

Thanks for the great work! I would like to use your repo, but I find it very difficult to get the compilation to work on my computer. I even tried to build from scratch on a clean cuda docker image. But still no luck. 

Is it possible for you to provide some kind of Docker support, be it Dockerfile or Docker image, I'm sure it will be greatly beneficial to a large number of developers like me to use your work.

kudo1026, 2021-06-22:
Alright, I finally get the compilation to work, and made a docker image on docker hub if anyone else would like to use.
The docker hub link: https://hub.docker.com/r/kudo1026/camera-calibration

closed 2021-06-22 by kudo1026

bigpo, 2021-07-29:
thanks for your docker support! however I encounter an error when i run this application:

> 09:02:28.136       cuda_buffer_inl.h:40    FATL| Cuda Error: CUDA driver version is insufficient for CUDA runtime version

could you please tell me how to solve it?

dbd99, 2024-04-24:
I have same issue with CUDA driver. Where you able to fix it? @bigpo