puzzlepaint/camera_calibration #51: CMake Error at CMakeLists.txt:118 (set_target_properties) “Can not find target to add properties to: genprebuilt”. What is the reason for this?

Ruiwen-qqq, 2021-07-27:
I have installed all the necessary dependencies. Then when I compile cmake, there will be such an error. How can this be solved? 
[image]

puzzlepaint, 2021-07-27:
I think that the CMake line from which this error originates is not relevant for compiling, just to organize the project in IDEs in a nicer way. While I don't know why this specific target from the error message seems to be missing in your case, as a workaround it should be fine to simply remove or comment out the line 118 in the top-level CMakeLists.txt file (or remove `genprebuilt` and any other targets that may be missing from it).

Ruiwen-qqq, 2021-07-28:
> I think that the CMake line from which this error originates is not relevant for compiling, just to organize the project in IDEs in a nicer way. While I don't know why this specific target from the [...]

I commented out this line, and then cmake succeeded. Thank you very much. By the way, I cmale this project under windows system.