puzzlepaint/camera_calibration #56: make[1]: *** [CMakeFiles/Makefile2:907: applications/camera_calibration/CMakeFiles/camera_calibration.dir/rule] Error 2 make: *** [Makefile:487: camera_calibration] Error 2

Barissevilmis, 2021-09-17:
I am trying to build a new library for camera calibration however get this error message when I use command make -j camera_calibration. Error occurs latest at 31% of the progress bar. I have checked all other libraries, cuda version and dependencies as well but they work fine.(CUDA 10.1)

Full error:
Scanning dependencies of target camera_calibration_apriltag_autogen
Scanning dependencies of target loguru
Scanning dependencies of target yaml-cpp2_autogen
Scanning dependencies of target genfiles
[  0%] Automatic MOC for target camera_calibration_apriltag
[  1%] Automatic MOC for target yaml-cpp2
[  2%] Generating pnglibconf.c
[  2%] Building CXX object CMakeFiles/loguru.dir/libvis/third_party/loguru/loguru.cpp.o
[  4%] Generating pngprefix.h
[  4%] Generating scripts/pnglibconf.c
[  4%] Generating scripts/symbols.out
options.awk: bad line (10): com
CMake Error at scripts/gensrc.cmake:68 (message):
  Failed to generate pnglibconf.tf5

make[3]: *** [libvis/third_party/libpng/CMakeFiles/genfiles.dir/build.make:85: libvis/third_party/libpng/pnglibconf.c] Error 1
make[3]: *** Waiting for unfinished jobs....
[  4%] Built target yaml-cpp2_autogen
options.awk: bad line (11): com
[  4%] Built target camera_calibration_apriltag_autogen
CMake Error at scripts/gensrc.cmake:41 (message):
  Failed to generate pnglibconf.tf7

make[3]: *** [libvis/third_party/libpng/CMakeFiles/genfiles.dir/build.make:108: libvis/third_party/libpng/scripts/pnglibconf.c] Error 1
make[2]: *** [CMakeFiles/Makefile2:776: libvis/third_party/libpng/CMakeFiles/genfiles.dir/all] Error 2
make[2]: *** Waiting for unfinished jobs....
Scanning dependencies of target yaml-cpp2
Scanning dependencies of target camera_calibration_apriltag
[  5%] Building CXX object applications/camera_calibration/third_party/yaml-cpp-0.6.0/CMakeFiles/yaml-cpp2.dir/src/convert.cpp.o
[  5%] Building CXX object applications/camera_calibration/third_party/yaml-cpp-0.6.0/CMakeFiles/yaml-cpp2.dir/yaml-cpp2_autogen/mocs_compilation.cpp.o
[  5%] Building CXX object applications/camera_calibration/third_party/yaml-cpp-0.6.0/CMakeFiles/yaml-cpp2.dir/src/directives.cpp.o
[  5%] Building CXX object applications/camera_calibration/third_party/yaml-cpp-0.6.0/CMakeFiles/yaml-cpp2.dir/src/emitfromevents.cpp.o
[  5%] Building CXX object applications/camera_calibration/third_party/yaml-cpp-0.6.0/CMakeFiles/yaml-cpp2.dir/src/binary.cpp.o
[  5%] Building CXX object applications/camera_calibration/CMakeFiles/camera_calibration_apriltag.dir/camera_calibration_apriltag_autogen/mocs_compilation.cpp.o
[  6%] Building CXX object applications/camera_calibration/third_party/yaml-cpp-0.6.0/CMakeFiles/yaml-cpp2.dir/src/emitterstate.cpp.o
[  6%] Building C object applications/camera_calibration/CMakeFiles/camera_calibration_apriltag.dir/third_party/apriltag/apriltag_pose.c.o
[  6%] Building CXX object applications/camera_calibration/third_party/yaml-cpp-0.6.0/CMakeFiles/yaml-cpp2.dir/src/emitter.cpp.o
[  8%] Building CXX object applications/camera_calibration/third_party/yaml-cpp-0.6.0/CMakeFiles/yaml-cpp2.dir/src/emit.cpp.o
[  8%] Building CXX object applications/camera_calibration/third_party/yaml-cpp-0.6.0/CMakeFiles/yaml-cpp2.dir/src/emitterutils.cpp.o
[  9%] Building C object applications/camera_calibration/CMakeFiles/camera_calibration_apriltag.dir/third_party/apriltag/apriltag.c.o
[  9%] Building C object applications/camera_calibration/CMakeFiles/camera_calibration_apriltag.dir/third_party/apriltag/apriltag_quad_thresh.c.o
[  9%] Building C object applications/camera_calibration/CMakeFiles/camera_calibration_apriltag.dir/third_party/apriltag/common/image_u8x3.c.o
[ 11%] Building C object applications/camera_calibration/CMakeFiles/camera_calibration_apriltag.dir/third_party/apriltag/common/image_u8x4.c.o
[ 11%] Building C object applications/camera_calibration/CMakeFiles/camera_calibration_apriltag.dir/third_party/apriltag/common/matd.c.o
[ 11%] Building C object applications/camera_calibration/CMakeFiles/camera_calibration_apriltag.dir/third_party/apriltag/common/pam.c.o
[ 11%] Building C object applications/camera_calibration/CMakeFiles/camera_calibration_apriltag.dir/third_party/apriltag/common/image_u8.c.o
[ 12%] Building C object applications/camera_calibration/CMakeFiles/camera_calibration_apriltag.dir/third_party/apriltag/common/g2d.c.o
[ 13%] Building C object applications/camera_calibration/CMakeFiles/camera_calibration_apriltag.dir/third_party/apriltag/common/pjpeg-idct.c.o
[ 13%] Building C object applications/camera_calibration/CMakeFiles/camera_calibration_apriltag.dir/third_party/apriltag/common/getopt.c.o
[ 13%] Building CXX object applications/camera_calibration/third_party/yaml-cpp-0.6.0/CMakeFiles/yaml-cpp2.dir/src/exceptions.cpp.o
[ 15%] Building C object applications/camera_calibration/CMakeFiles/camera_calibration_apriltag.dir/third_party/apriltag/common/homography.c.o
[ 15%] Building C object applications/camera_calibration/CMakeFiles/camera_calibration_apriltag.dir/third_party/apriltag/common/pjpeg.c.o
[ 15%] Building CXX object applications/camera_calibration/third_party/yaml-cpp-0.6.0/CMakeFiles/yaml-cpp2.dir/src/memory.cpp.o
[ 16%] Building CXX object applications/camera_calibration/third_party/yaml-cpp-0.6.0/CMakeFiles/yaml-cpp2.dir/src/exp.cpp.o
[ 16%] Building CXX object applications/camera_calibration/third_party/yaml-cpp-0.6.0/CMakeFiles/yaml-cpp2.dir/src/node.cpp.o
[ 18%] Building CXX object applications/camera_calibration/third_party/yaml-cpp-0.6.0/CMakeFiles/yaml-cpp2.dir/src/node_data.cpp.o
[ 18%] Building CXX object applications/camera_calibration/third_party/yaml-cpp-0.6.0/CMakeFiles/yaml-cpp2.dir/src/nodebuilder.cpp.o
[ 18%] Building C object applications/camera_calibration/CMakeFiles/camera_calibration_apriltag.dir/third_party/apriltag/common/svd22.c.o
[ 18%] Building C object applications/camera_calibration/CMakeFiles/camera_calibration_apriltag.dir/third_party/apriltag/common/pnm.c.o
[ 18%] Building CXX object applications/camera_calibration/third_party/yaml-cpp-0.6.0/CMakeFiles/yaml-cpp2.dir/src/nodeevents.cpp.o
[ 18%] Building C object applications/camera_calibration/CMakeFiles/camera_calibration_apriltag.dir/third_party/apriltag/common/time_util.c.o
[ 19%] Building C object applications/camera_calibration/CMakeFiles/camera_calibration_apriltag.dir/third_party/apriltag/common/string_util.c.o
[ 20%] Building CXX object applications/camera_calibration/third_party/yaml-cpp-0.6.0/CMakeFiles/yaml-cpp2.dir/src/null.cpp.o
[ 20%] Building CXX object applications/camera_calibration/third_party/yaml-cpp-0.6.0/CMakeFiles/yaml-cpp2.dir/src/ostream_wrapper.cpp.o
[ 22%] Building C object applications/camera_calibration/CMakeFiles/camera_calibration_apriltag.dir/third_party/apriltag/common/unionfind.c.o
[ 22%] Building C object applications/camera_calibration/CMakeFiles/camera_calibration_apriltag.dir/third_party/apriltag/common/workerpool.c.o
[ 22%] Building C object applications/camera_calibration/CMakeFiles/camera_calibration_apriltag.dir/third_party/apriltag/common/zhash.c.o
[ 22%] Building CXX object applications/camera_calibration/third_party/yaml-cpp-0.6.0/CMakeFiles/yaml-cpp2.dir/src/parser.cpp.o
[ 22%] Building C object applications/camera_calibration/CMakeFiles/camera_calibration_apriltag.dir/third_party/apriltag/common/zmaxheap.c.o
[ 23%] Building CXX object applications/camera_calibration/third_party/yaml-cpp-0.6.0/CMakeFiles/yaml-cpp2.dir/src/parse.cpp.o
[ 25%] Building C object applications/camera_calibration/CMakeFiles/camera_calibration_apriltag.dir/third_party/apriltag/common/zarray.c.o
[ 25%] Building CXX object applications/camera_calibration/third_party/yaml-cpp-0.6.0/CMakeFiles/yaml-cpp2.dir/src/regex_yaml.cpp.o
[ 26%] Building C object applications/camera_calibration/CMakeFiles/camera_calibration_apriltag.dir/third_party/apriltag/tag36h11.c.o
[ 27%] Building CXX object applications/camera_calibration/third_party/yaml-cpp-0.6.0/CMakeFiles/yaml-cpp2.dir/src/scanner.cpp.o
[ 29%] Building CXX object applications/camera_calibration/third_party/yaml-cpp-0.6.0/CMakeFiles/yaml-cpp2.dir/src/scantoken.cpp.o
[ 29%] Building CXX object applications/camera_calibration/third_party/yaml-cpp-0.6.0/CMakeFiles/yaml-cpp2.dir/src/scanscalar.cpp.o
[ 29%] Building CXX object applications/camera_calibration/third_party/yaml-cpp-0.6.0/CMakeFiles/yaml-cpp2.dir/src/singledocparser.cpp.o
[... cut ...]
[ 31%] Building CXX object applications/camera_calibration/third_party/yaml-cpp-0.6.0/CMakeFiles/yaml-cpp2.dir/src/contrib/graphbuilderadapter.cpp.o
[ 31%] Linking CXX static library libcamera_calibration_apriltag.a
[ 31%] Built target camera_calibration_apriltag
[ 31%] Linking CXX static library libyaml-cpp2.a
[ 31%] Built target yaml-cpp2
[ 31%] Linking CXX shared library libloguru.so
[ 31%] Built target loguru
make[1]: *** [CMakeFiles/Makefile2:907: applications/camera_calibration/CMakeFiles/camera_calibration.dir/rule] Error 2
make: *** [Makefile:487: camera_calibration] Error 2

puzzlepaint, 2021-09-18:
The relevant error output is:
```
options.awk: bad line (10): com
CMake Error at scripts/gensrc.cmake:68 (message):
Failed to generate pnglibconf.tf5

make[3]: *** [libvis/third_party/libpng/CMakeFiles/genfiles.dir/build.make:85: libvis/third_party/libpng/pnglibconf.c] Error 1
make[3]: *** Waiting for unfinished jobs....

options.awk: bad line (11): com
CMake Error at scripts/gensrc.cmake:41 (message):
Failed to generate pnglibconf.tf7
```

These errors come from the third-party `libpng` library that is included in the source. For example, one error is output from this line:
https://github.com/puzzlepaint/camera_calibration/blob/846bb5282a23b68e9fd7f3d240f7c979a016dea6/libvis/third_party/libpng/scripts/gensrc.cmake.in#L41

I would guess that the error is not directly related to the camera_calibration project. Googling the error message for example brought up this, where the same error message occurred:
https://stackoverflow.com/questions/59547342/libpng-building-issue-on-windows-10

In that case, this was resolved by changing the line endings.

Barissevilmis, 2021-09-27:
Thank you, after changing the line endings problem was fixed.

closed 2021-09-27 by Barissevilmis