puzzlepaint/camera_calibration #63: feature extraction and .bin

liangshunkun, 2022-05-13:
Hi @[puzzlepaint](https://github.com/puzzlepaint), I am very interested in your work on calibration pattern feature extraction. When I run the code for feature extraction I get a “.bin” and I would like to know how the ones inside represent a description of a feature point. Does the file indicate the position of the feature points on the calibration plate or on the image?
In additional, I want to find the same feature points in two images that contain a calibration pattern. how can get that?

puzzlepaint, 2022-05-14:
For the format of the .bin files, please see the code that is used to write them, starting here:
https://github.com/puzzlepaint/camera_calibration/blob/32173589ae4db7491ace3173b5bdf2bf65f98716/applications/camera_calibration/src/camera_calibration/io/calibration_io.cc#L51

Among other data, this file contains the feature positions in the images, as written here:
https://github.com/puzzlepaint/camera_calibration/blob/32173589ae4db7491ace3173b5bdf2bf65f98716/applications/camera_calibration/src/camera_calibration/io/calibration_io.cc#L105

In order to detect the same feature points in different images, their feature ID can be compared, which is written here:
https://github.com/puzzlepaint/camera_calibration/blob/32173589ae4db7491ace3173b5bdf2bf65f98716/applications/camera_calibration/src/camera_calibration/io/calibration_io.cc#L108

The feature positions on the calibration plate(s) are given by the "known geometries", which are written below:
https://github.com/puzzlepaint/camera_calibration/blob/32173589ae4db7491ace3173b5bdf2bf65f98716/applications/camera_calibration/src/camera_calibration/io/calibration_io.cc#L113

liangshunkun, 2022-07-11:
Thank you very much！！ I think I ' ve figured it out

closed 2022-07-11 by liangshunkun