puzzlepaint/camera_calibration #65: Convergence issue

boazsaips, 2022-09-28:
Hi,

I'm trying to calibrate a central-generic model to my iPhone camera (as a test). I took ~400 photos of the calibration board, with a good coverage of almost all pixels, but the error histogram I get at the end of the calibration process does not look good:

[image]

The image resolution is 4032x3024 and the I set the CELL_SIZE to 30. I tried CELL_SIZE=80 but the histogram remained quite similar.
Am I missing something? your advice will be appreciated.

Boaz

puzzlepaint, 2022-09-28:
Hi, some things to check could be:
* Is any issue visible in the other visualizations of the calibration result?
* What is the clarity of the pictures when zooming in to the pattern features? In case the centers of the star features aren't sharp, a larger spread of the reprojection errors, as in your result, would be expected.
* If the camera uses a rolling shutter, and you took the pictures with the iPhone held in your hand, how much does it affect the result? I imagine that pressing the button to take a picture leads to some small shaking of the device, and the rotational part of that might cause relevant distortions for rolling-shutter cameras.

boazsaips, 2022-10-03:
Apparently, the problem was the autofocus mechanism of the iPhone camera. I took ~50 pictures again, with the autofocus disabled, and the optimization converged nicely.

closed 2022-10-03 by boazsaips