puzzlepaint/camera_calibration #66: Matching-based refinement rendering issues

seeeheng, 2023-02-18:
Thank you for this amazing calibration toolbox! I am currently evaluating both the matching-based and symmetry-based refinement techniques and have managed to get symmetry-based refinement working to great success.

However, matching-based refinement tends to give unstable results.

In turning on the commented out portions of the code for debugging, I find that these [lines](https://github.com/puzzlepaint/camera_calibration/blob/32173589ae4db7491ace3173b5bdf2bf65f98716/applications/camera_calibration/src/camera_calibration/feature_detection/cpu_refinement_by_matching.h#L232-L261) (also reproduced below) does not accurately render the star pattern when used with my custom target, but works well on the sample dataset provided.

I was wondering if you would have any insight as to why this is happening? Thank you!

```python// The positions are specified in pixel-center origin convention.
template <typename T, typename DerivedA, typename DerivedB>
bool RefineFeatureByMatching(
    int num_samples, //ok
    const vector<Vec2f>& samples, //ok
    const Image<T>& image, //ok
    int window_half_size, //ok
    const MatrixBase<DerivedA>& position, //*
    const MatrixBase<DerivedB>& local_pattern_tr_pixel, //ok
    const PatternData& pattern, //*
    Vec2f* out_position, //ok
    float* final_cost, //ok
    bool debug) {
  constexpr int kNumAntiAliasSamples = 16;
  // Using the local homography, render the known pattern at the samples. Each
  // sample defines a pixel offset from the feature position.
  vector<float> rendered_samples(num_samples);
  for (int i = 0; i < num_samples; ++ i) {
    float sum = 0;
    for (int s = 0; s < kNumAntiAliasSamples; ++ s) {
      // Samples spread in [-0.5, 0.5], i.e., within the range of one pixel.
      Vec2f pixel_offset = window_half_size * samples[i] +
                           Vec2f(-0.5 + 1 / 8.f + 1 / 4.f * (s % 4),
                                 -0.5 + 1 / 8.f + 1 / 4.f * (s / 4));
      Vec2f pattern_offset = Vec3f(local_pattern_tr_pixel * pixel_offset.homogeneous()).hnormalized();
      sum += pattern.PatternIntensityAt(pattern_offset);
    }
    // Normalization by kNumSubpixelSamples is not necessary here since an
    // affine intensity transformation is optimized for later.
    rendered_samples[i] = sum;
  }
```
Rendered image: (from my dataset)
[image]

Expected render: (from your dataset)
[image]

seeeheng, 2023-02-18:
I am attempting to render the same pattern (8 star segments) as the one provided in your sample dataset.

seeeheng, 2023-02-19:
I realized, in my custom implementation, that I forgot to account for the fact that PatternIntensityAt requires coordinates from [-0.5, 0.5], with pixel center convention. After adding the necessary transform, I am able to render the pattern correctly.

[image]

puzzlepaint, 2023-02-21:
Thanks for the update - can this issue be closed, or are the results still unstable after the fix?

seeeheng, 2023-02-22:
Actually I did have more questions about the results.

I've un-commented all the debug code [here](https://github.com/puzzlepaint/camera_calibration/blob/32173589ae4db7491ace3173b5bdf2bf65f98716/applications/camera_calibration/src/camera_calibration/feature_detection/cpu_refinement_by_matching.h#L263-L291) and [here](https://github.com/puzzlepaint/camera_calibration/blob/32173589ae4db7491ace3173b5bdf2bf65f98716/applications/camera_calibration/src/camera_calibration/feature_detection/cpu_refinement_by_matching.h#L162-L195) and all the rendered and transformed patterns work fine.

I've added statements to track loss and while the intensity-based refinement appears to be converging (loss reduces significantly with each iteration, eventually "converges", in my example from 41181.1 to 3393.69), the final results differ wildly from the symmetry-based method.

I've included the features refined from both methods below, where the red signifies results from the intensity based matching, and green signifies results from symmetry.

[image]

As same predictions and transformations are used between both methods, do you have any insight into why the output from the methods would differ by such a huge degree? 

Steps I have tried:
1. Changing anti-aliasing sample # 
2. Changing window_half_extent values

Thank you so much for your time!

puzzlepaint, 2023-02-23:
I quickly verified on the sample dataset that the non-CUDA version of the intensity based refinement generally does seem to work in the current state of the code. The following image shows a typical result from there, where the red point is the final result, the white point is the result after intensity-based refinement, and the gray point is the initial prediction:

[image]

It has been a while that I worked on this, but this matches with what I remember as the typical difference between the two methods. Nevertheless, the difference might be larger on other datasets. The main possible reason that I can think of at the moment is the following:

The implementation of the intensity-based refinement does not optimize the whole homography that it uses, instead it optimizes for a translation only. Perhaps the predicted homography is inaccurate in your dataset, such that optimizing the translation is not sufficient? The symmetry-based refinement, on the other hand, optimizes the whole homography, so it can "fix" an inaccurate initial guess of the homography.

seeeheng, 2023-02-24:
You're right - the matching based refinement was working well, but as it's only optimizing for the translation shift, and the original homography was off, it predicts a feature that isn't accurate.

In my pipeline I feed the homography predicted from the symmetry based refinement into the matching based refinement and the results look great. Thank you so much, I really appreciate the time and effort you took to help me with this.

closed 2023-02-24 by seeeheng