puzzlepaint/camera_calibration #69: Issue with building

xiaomianzhou, 2023-05-25:
Hi,
I have already cmake successfully, but when I try to make, I got the following error
`[ 95%] Building CXX object applications/camera_calibration/CMakeFiles/camera_calibration_baselib.dir/src/camera_calibration/calibration_report.cc.o
[ 97%] Building CXX object applications/camera_calibration/CMakeFiles/camera_calibration_baselib.dir/src/camera_calibration/dataset.cc.o
[ 97%] Building CXX object applications/camera_calibration/CMakeFiles/camera_calibration_baselib.dir/src/camera_calibration/util.cc.o
[ 97%] Linking CXX shared library libcamera_calibration_baselib.so
/usr/bin/ld: /usr/lib/libboost_filesystem.a(operations.o): relocation R_X86_64_PC32 against symbol `_ZSt7nothrow@@GLIBCXX_3.4' can not be used when making a shared object; recompile with -fPIC
/usr/bin/ld: final link failed: bad value
collect2: error: ld returned 1 exit status
make[3]: *** [applications/camera_calibration/CMakeFiles/camera_calibration_baselib.dir/build.make:838: applications/camera_calibration/libcamera_calibration_baselib.so] Error 1
make[2]: *** [CMakeFiles/Makefile2:832: applications/camera_calibration/CMakeFiles/camera_calibration_baselib.dir/all] Error 2
make[1]: *** [CMakeFiles/Makefile2:872: applications/camera_calibration/CMakeFiles/camera_calibration.dir/rule] Error 2
make: *** [Makefile:491: camera_calibration] Error 2
`
Does anyone have any advice on how to fix this?
Thanks a lot!

xiaomianzhou, 2023-05-26:
I tried the following and the project is built successfully, maybe one of these things worked.

I rebuild boost with -fPIC.
```shell
./b2 cxxflags=-fPIC cflags=-fPIC
``` 

I add the following to CMakeLists.txt
```shell
set(CMAKE_CXX_FLAGS "-Wall -fPIC -std=c++11")
```

spdfghi, 2024-10-19:
@xiaomianzhou 兄弟，能联系下吗