puzzlepaint/camera_calibration #75: Central OpenCV gives wrong results

LulaSan, 2024-08-23:
Hello, I am trying to use the results obtained from the opencv model and they do not generate good results.
Using the following code
```
    cv::Mat cameraMatrix = (cv::Mat_<double>(3, 3) << parameters[0], 0, parameters[2], 0, parameters[1], parameters[3], 0, 0, 1);
    // Create the distortion coefficients
    cv::Mat distCoeffs = (cv::Mat_<double>(5, 1) << parameters[4], parameters[5], parameters[10], parameters[11], parameters[6]);
    // Rectify the image
    cv::Mat rectifiedImage;
     cv::undistort(inputImage, rectifiedImage, cameraMatrix, distCoeffs);
```
it does not give good results, why?

sscholbe, 2024-08-24:
Hi! The OpenCV model in this library uses a rational where the radial distortion factor is modeled as

$$\frac{1+k_1 r^2+k_2 r^4+k_3 r^6}{1+k_4 r^2+k_5 r^4+k_6 r^6}$$

So you must include $k_4,k_5,k_6$ in your distCoeffs and pass the `CALIB_RATIONAL_MODEL` flag to undistort.

LulaSan, 2024-08-26:
Hi @sscholbe   thank you for your answer.
In the documentation, https://docs.opencv.org/4.x/d9/d0c/group__calib3d.html#ga69f2545a8b62a6b0fc2ee060dc30559d, the distorsion coefficient is : 
[image]

What I am currently using from the output vector is k1,k2,p1,p2,k3.
If I use also k4,k5,k6 the result is not different.
original image from author dataset:
[image]
Rectified Image:
[image]

Intrisics0.yaml
```
type : CentralOpenCVModel
width : 640
height : 480
parameters : [277.3961936503, 277.25286238918, 307.7938365571, 233.0473084845, 12.03989608475, 5.8339642150155, 0.18887783289071, 12.356366463215, 9.7142850536517, 1.1166962340432, -0.00027457277380875, 5.9557517223146e-05]
```
Histogram
[image]

sscholbe, 2024-08-27:
Hi, maybe you messed up the order of the coefficients somewhere. The library exports them in order
$$(f_x,f_y,c_x,c_y,k_1,k_2,k_3,k_4,k_5,k_6,p_1,p_2)$$
OpenCV needs the distortion coefficients in order
$$(k_1,k_2,p_1,p_2,k_3,k_4,k_5,k_6)$$

The following code undistorts the image correctly:
```python
import cv2
import numpy as np

# fx, fy, cx, cy, k1, k2, k3, k4, k5, k6, p1, p2
# copied from you
parameters = [277.3961936503, 277.25286238918, 307.7938365571, 233.0473084845, 12.03989608475, 5.8339642150155,
              0.18887783289071, 12.356366463215, 9.7142850536517, 1.1166962340432, -0.00027457277380875, 5.9557517223146e-05]

img = cv2.imread("input.png")
mtx = np.asarray([[parameters[0], 0, parameters[2]],
                  [0, parameters[1], parameters[3]],
                  [0, 0, 1]], float)

# k1, k2, p1, p2, k3, k4, k5, k6
dist = np.asarray([parameters[4], parameters[5], parameters[10], parameters[11],
                  parameters[6], parameters[7], parameters[8], parameters[9]], float)

rectified = cv2.undistort(img, mtx, dist)

cv2.imwrite("output.png", rectified)
```

Result:
[image]

Your lens also has quite heavy distortion. You should read into `getOptimalNewCameraMatrix` to find a pinhole that doesn't lose as many pixels as you do right now (which you can pass via `newCameraMtx` to `undistort`).

LulaSan, 2024-10-11:
Thank you very much! That worked.

closed 2024-10-11 by LulaSan